MobiFlight/mobiflight-panels #90: Illumination

Krakend70, 2022-02-14:
Hi
I am discovering your work - it is excellent. Thanks !!!
Now, I am strating by coincidence in the A32X Illumination folder and the components are a bit strnge: For ex., led for backlight are... red ! Is that normal ? Do I have to check all components ?

Many thanks again !

ssewell, 2022-02-14:
@Krakend70 The color of the LED to be used should be indicated in the Name field of the BOM
[image]

The LEDs included in the EasyEDA Common Library are somewhat limited, so most of the representations are going to use that generic model (you'll see it as such when viewing a 3D representation of the PCB) and will typically include a "RED" in the footprint, but you should reference the Name field to determine the actual color to be used.

It's been lower on my priorities, but I intend to create a symbol/footprint library that uses a common that doesn't include a color name in the footprint itself.

ssewell, 2022-02-16:
We welcome any contributions or corrections! 

In our current workflow, the EasyEDA JSON files are artifacts of the shared workspace in EasyEDA. Unfortunately, the exported JSON files don't lend themselves well to the current diff mechanism in GitHub. The [MobiFlight Discord server](https://discord.gg/GY5e4h9T) is a great way to meet up and inquire about gaining access to the MobiFlight EasyEDA team. The #pcb-design channel is fairly active.

Alternatively, you can submit a PR including a description of the changes in the comments (to assist in identifying the changes) and it will be merged into the main branch, post review.

12V will be labeled with a Net Port and 5V will be labeled with either a Net Port or Net Flag (+5V), with varying names, depending on who created the design. Most designs will have a USB input via the on-board Arduino and a 12V input if backlighting is used.

Krakend70, 2022-02-16:
OK thanks for the infos
Probably last question about ECAM: Where are the 'External for Switching Panel' headers supposed to be connected to ? Where can I find the info ? Thx

Krakend70, 2022-02-17:
About my last post, I mean the following: Are these headers to be connected with the 'Pedestal Switching Panel', correct ? - By 'Pedestal Switching Panel', I mean the panel highlighted in the attached picture. In that case, do you have lraedy drafted such panel and PCB ?
[image]

ssewell, 2022-02-18:
The ECAM panel provides an interface for three pins per rotary switch on the Switching panel, with the intent of a common GND being shared among the two panels. However, the Switching panel hasn't been developed yet.

Krakend70, 2022-02-18:
got it thx

closed 2022-04-19 by ssewell