MobiFlight/mobiflight-panels #134: Some Questions with with the

KMC-Workshop2024, 2024-12-09:


KMC-Workshop2024, 2024-12-09:
It will be better if there is a screen size details add in README.

datamann, 2024-12-09:
You can create it and make a pr.

KMC-Workshop2024, 2024-12-09:
> You can create it and make a pr.

Hmmmm if I know the size of the screen, I will do that :)
but actually I don't really know what model of the screen I should buy 

Could you tell me what model of screen I should buy? Thanks a lot :)

KMC-Workshop2024, 2024-12-09:
> You can create it and make a pr.

It would be even better if you could guide me in making the MCDU, and provide a list of your raw materials when making the MCDU. :)

Here is my email address:george20070301@gmail.com 

Hope you have a nice day.

datamann, 2024-12-09:
I used this screen for my MCDU
https://www.aliexpress.com/item/1005005394656264.html

closed 2024-12-20 by KMC-Workshop2024