MobiFlight/mobiflight-panels #135: What should I choose for the TCAS select switch？

KMC-Workshop2024, 2024-12-21:
It will be better if someone can upload some details about the TCAS select rotate switch :) 

I have search for the switch for a long time, but I don't know how to choose. I found some switch on Aliexpress, but it seems like not suitable for the TCAS panel, and the rotation angle is incorrect.

Can someone help me?

Thanks a lot    :)

datamann, 2024-12-21:
Here is one option for you

https://www.cgtrader.com/3d-print-models/hobby-diy/electronics/airbus-a320-wxr-tcas-and-chrono-knobs

KMC-Workshop2024, 2024-12-21:
> Here is one option for you
> 
> https://www.cgtrader.com/3d-print-models/hobby-diy/electronics/airbus-a320-wxr-tcas-and-chrono-knobs

I mean the components that are soldered to the PCB, not the 3d model :)
Can you give me a link for the components?
[image]

closed 2025-08-13 by KMC-Workshop2024